OxfordIonTrapGroup/stabilizer_current_sense #2: 1v.0 rc1 design review

hartytp, 2019-04-08:
@gkasprow I've just posted a first version of the design, would you mind taking a look?

The only obvious issue I see is #1 

Note that I've put quite a lot of thought into the grounding, since we need to measure <1uV voltages across the sense resistor. In a previous version we had some nasty issues like: an LED that flashed at ~1Hz and we could see a corresponding signal on the current measurement; the current sunk by the shunts created a ground voltage that altered the current measurement. These were all fixed with grounding, but it took some care!

Other than that, it would be great if you could have a look at the usuals (schematic, layout, mechanics) and, since I'm new to Altium, do let me know if you see anything silly with the way I've set the Altium project up.

Otherwise, I'll order some of these boards in the next few days. Is one enough for you?

hartytp, 2019-04-09:
Now on rc2.

Main change is that when I added a mating connector for the terminal block I noticed that the locking terminal blocks wouldn't fit. Fixed the component placements and panels to allow use of locking terminal blocks.

NB we won't be able to use one of the fancy Schroff panels with EMC gaskets unless we trim the pins on the terminal block. But, the same goes for the SMAs on Stabilizer AFAICT.

Other than that, the mechanics looks good apart from #1.

gkasprow, 2019-04-10:
@hartytp 
- trimming pins is the usual procedure.
- there are missing decoupling caps on IC11 and IC10 supplies
- move all 3D models to layer 13
- missing board outline on mechanical 1. This is needed for PCB milling.
- mask vias, don't treat them as test points because can be easily damaged with a probe

gkasprow, 2019-04-10:
edit, the decoupling caps are there, on other sheet :)

gkasprow, 2019-04-10:
What is D1 for? Wouldn't it be better to use it in the feedback loop to close it for negative inputs?

hartytp, 2019-04-10:
> trimming pins is the usual procedure

ack. Is that something that you specify somewhere on the PCB? Or just leave to the assembly house to figure out?

> there are missing decoupling caps on IC11 and IC10 supplies
> edit, the decoupling caps are there, on other sheet :)

Nope, they're there on the same sheet. C34-C36

> move all 3D models to layer 13

Done

> missing board outline on mechanical 1. This is needed for PCB milling.

Okay, I'll add that.

> mask vias, don't treat them as test points because can be easily damaged with a probe

Done. Thanks

hartytp, 2019-04-10:
> What is D1 for? Wouldn't it be better to use it in the feedback loop to close it for negative inputs?

May not be necessary. The idea is to protect the transitor's base from negative voltages when the feedback loop rails low.

gkasprow, 2019-04-10:
the air flow cooling the output transistors can be blocked by the DC/DC converter. I don't know how hot they are going to be.
Trimming of the pins will be done during panel assembly so the assembly house will figure it out

hartytp, 2019-04-10:
> the air flow cooling the output transistors can be blocked by the DC/DC converter. I don't know how hot they are going to be.

Imax is about 0.1A. The power depends on the coil voltage. Assume a max coil voltage of something like 10V (which is pretty high) the max power will be 1 W. The [heatsink](https://www.fischerelektronik.de/web_fischer/en_GB/heatsinks/C02/Attachable%20heatsink/PG/FK242SA220O/search.xhtml) specifies 26K/W thermal resistance, but not sure what air flow that assumes. Either way, the transistors shouldn't get too hot, maybe 50C or so.

I'm assuming we'll mount the circuit virtically in an enclosure with some vent holes on the top.

gkasprow, 2019-04-10:
When making cuts via power planes, make sure they are not shorted on other layers for AC.
At the moment the cutout is shorted by R50. The N15 plane is also placed across the cutout.

gkasprow, 2019-04-10:
the EMI filter should be placed as close to the connector as possible. When placed somewhere else its effectiveness gets degraded, especially for higher frequencies. it would fit nicely on the bottom side, close to the connector.

gkasprow, 2019-04-10:
watch the via sizes on power rails. I add a dedicated rule and make sure all power rails are added to the power class.
Use two or three vias in parallel. If Altium is removing them, add piece of copper fill to connect them together (p-f command). At the moment the pin 1 of EMI filter is connected  to the power connector witn big via which has 0.5mm hole

gkasprow, 2019-04-10:
Wouldn't make it better to place RN1 on the bottom side? The feedback traces seem to be quite long.
I don't care on signal integrity but interference pickup.

gkasprow, 2019-04-10:
Good idea with adding panels to the BOM!

gkasprow, 2019-04-10:
are you sure the connectors you used are good for such stacking? On the 3D view it looks like the header may have too short pins to ensure reliable contact. It depends on how the receptacles are built - if they have an active contact surface close to the pin entry or at some depth.

hartytp, 2019-04-10:
Thanks Greg. Let me know when you've finished your review.

gkasprow, 2019-04-10:
I didn't found any other issues.

hartytp, 2019-04-10:
Thanks @gkasprow!

hartytp, 2019-04-10:
> move all 3D models to layer 13

So, here you mean J7 as well as the screws, washers and stand-offs, right? I think everything else was already on the correct layer.

For stand-off/alignment purposes, is mechanical 1 the same as top layer?

hartytp, 2019-04-10:
> missing board outline on mechanical 1. This is needed for PCB milling.

@gkasprow is there an easy way to have altium copy the board outline on to m1 or do I need to manually trace it using zero-thickness lines?

hartytp, 2019-04-10:
> When making cuts via power planes, make sure they are not shorted on other layers for AC.
At the moment the cutout is shorted by R50.

R50 and other similar resistors are DNP by default. They are deliberately placed there to short across the isolation gap if we decide we want to tie all grounds together at some point. There is a schematic annotation about this.

> The N15 plane is also placed across the cutout.

Several power planes cut across the isolation gap. But, this is a requirement to bring the signals out. I don't think we can avoid it. I wasn't expecting this to be a problem, as the capacitance between the layers should be relatively small and the noise frequencies should be relatively low. Do you think this is really likely to be an issue? How would you alter the layout to improve it?

> the EMI filter should be placed as close to the connector as possible. When placed somewhere else its effectiveness gets degraded, especially for higher frequencies. it would fit nicely on the bottom side, close to the connector.

I did think about doing that. However, in this particular case, I deliberately chose not to do that. The reason is that the area near the barrel connector has my low-noise AFE ground, so I do not want to ground the CMC there. I want to ground the CMC in the "noisy" region near the main SMPS.

I know this will degrade the CMC performance somewhat for high frequenices, but my assumption was that the bulk of the noise we want to filter out is <1MHz so the coupling between the power input traces and the surrounding ground planes should be very small.

What do you think?

> watch the via sizes on power rails. I add a dedicated rule and make sure all power rails are added to the power class.
Use two or three vias in parallel. If Altium is removing them, add piece of copper fill to connect them together (p-f command). At the moment the pin 1 of EMI filter is connected to the power connector witn big via which has 0.5mm hole

> Wouldn't make it better to place RN1 on the bottom side? The feedback traces seem to be quite long.
I don't care on signal integrity but interference pickup.

Yes, I guess we can do that. I'm used to hand soldering PCBs so I like to keep all components on one side where possible :) It's also nice to keep all components on the same layer in case we want to add a screening around the low noise parts can or something later on. Also, the layout currently puts keeps all critical ground connections (the reference, the dividers, the DAC and the feedback resistors) very close together, which seemed nice.

In any case, the feedback traces are not so long (~15mm) and the subsequent stages have filtering.

So, all in all, I hoped the current layout would be fine. What do you think?

> Are you sure the connectors you used are good for such stacking? On the 3D view it looks like the header may have too short pins to ensure reliable contact. It depends on how the receptacles are built - if they have an active contact surface close to the pin entry or at some depth.

Yes...this is https://github.com/OxfordIonTrapGroup/stabilizer_current_sense/issues/1 I think it may be an issue.

Let's discuss the problem in that thread. Briefly though:
- there is no connector that I could find which is a closer match for our stack up. The longer connectors are about 1mm too long to mate with the headers on Stabilizer.
- one possibility would be to solder the connectors slightly below the AFE PCB to increase the mating lenght. e.g. we can solder them in situ with the board mounted on Stabilizer, or we can add a 1mm spacer before soldering.
- From the connector models in the step file, it looks like the female connectors are sprun-loaded with a contact point right at the tip, so this may not be an issue in the end. But, it's hard to be confident until we play with the hw...

gkasprow, 2019-04-10:
> When making cuts via power planes, make sure they are not shorted on other layers for AC.
>     At the moment the cutout is shorted by R50.
>
> R50 and other similar resistors are DNP by default. They [...]

OK
> The N15 plane is also placed across the cutout.
>
> Several power planes cut across the isolation gap. But, this is a requirement to bring the signals out. I don't think we can avoid it. I wasn't [...]

I would simply make some supply rails longer that go around the slot. Look how it was made on Sampler.

> the EMI filter should be placed as close to the connector as possible. When placed somewhere else its effectiveness gets degraded, especially for higher frequencies. it would fit nicely on the [...]

you can ground it in another place. At the moment the input traces are capacitively coupled with your AFE ground.

> 
> I know this will degrade the CMC performance somewhat for high frequenices, but my assumption was that the bulk of the noise we want to filter out is <1MHz so the coupling between the power input traces and the surrounding ground planes should be very small.
> 
> What do you think?

probably does not really mater at such low frequencies we are working.

> watch the via sizes on power rails. I add a dedicated rule and make sure all power rails are added to the power class.
>     Use two or three vias in parallel. If Altium is removing them, add piece of [...]

that is fine, I'm talking only on general engineering rules :)

> Are you sure the connectors you used are good for such stacking? On the 3D view it looks like the header may have too short pins to ensure reliable contact. It depends on how the receptacles are [...]

this needs to be communicated to the assembly house. They will use standoffs to mount the connector in elevated way.

>     From the connector models in the step file, it looks like the female connectors are sprun-loaded with a contact point right at the tip, so this may not be an issue in the end. But, it's hard to be confident until we play with the hw...
> 

this should not be an issue then.

hartytp, 2019-04-11:
@gkasprow for power planes, what via hole and annuls size do you recommend? How many in parallel for a few hundred mA?

closed 2019-04-11 by hartytp

gkasprow, 2019-04-11:
0.4/0.7mm, 2 in parallel.