OxfordIonTrapGroup/stabilizer_current_sense #8: v1.0 testing

hartytp, 2019-10-16:
- [ ] measure all voltage rails and check within spec. Check with scope that nothing oscillating etc
- [ ] program SPI AFE DAC and verify output
- [ ] short current input and measure noise spectrum (if we can't do this because of offsets, we can use a battery + precision resistor to produce a small bias current with very low noise)
- [ ] Compare noise measurements to simulations in notebook
- [ ] Get stabilizer CPU DAC working to bias current shunts
- [ ] test current shunts by hooking up to external current source and putting a square-wave on the stabilizer SPI DAC. Look at sense resistor on scope to measure step response
- [ ] try running closed-loop

pathfinder49, 2019-10-16:
| V Rail    | Voltage (DMM) |
|------------|---------------------|
| P15V0   | 15.12 V |
| N15V0   | -14.96 V |
| P12V0A | 12.01 V |
| N12V0A | -12.05 V |
| 12V0AS | 12.29 V |
| VREF    | 2.500 V |
| P5V0    | 4.971 V |

Verified on a scope that the rails aren't oscillating.

hartytp, 2019-10-16:
Are those all within spec for the PCB?

pathfinder49, 2019-10-16:
> 
> 
> Are those all within spec for the PCB?

Those are all within spec of the connected components. I can't seem to find a tolerance within the PDF.

hartytp, 2019-10-16:
> Those are all within spec of the connected components

That's all I meant; cast an eye over the PCB, dig out the ICs and see what accuracy you expect from that.

pathfinder49, 2019-10-28:
Below are the stabilizer ADC readings when connected to stabilizer_current_sense. (I've plotted every 100th point)

`np.diff(data).std()/2**.5` = 10.8 LSB or 1.3 mV

[image]
[image]

pathfinder49, 2019-10-28:
With R33 and R34 populated and Coil COM and Coil LOW shorted.

[image]
[image]

There is a significant voltage offset with this configuration. See [new issue](https://github.com/OxfordIonTrapGroup/stabilizer_current_sense/issues/9).

pathfinder49, 2019-10-28:
The voltage fluctuations appear to originate from a [broken capacitor in the DAC filter](https://github.com/OxfordIonTrapGroup/stabilizer_current_sense/issues/10).

Using a different current_sense_board:
`np.diff(data).std()/2**.5` = 10.8 LSB or 1.3 mV (no significant change to previous board)
(6.2 mV RMS noise on scope)
[image]
[image]

hartytp, 2019-10-28:
@pathfinder49 can you fft that to extract the nsd and compare it to the simulation please? Looks about right but would be nice to see the nsd plot

hartytp, 2019-10-28:
@pathfinder49 are you saying the ADC reads 1.3mVrms but the scope gives 6.4?

pathfinder49, 2019-10-28:
Yes, I'll double check my conversion once I've eaten. Might have /2 instead of x2...

pathfinder49, 2019-10-28:
I believe my conversion is correct. The `np.diff(data).std()/2**.5` disagrees with the rms of the ADC data (2.4 mV). However, this RMS also disagrees with the scope...

Below is the NSD of the ADC data (assuming 1 MHz sampling TBC):
[image]

hartytp, 2019-10-28:
I'm confused. Why are you differentiating the data?

hartytp, 2019-10-28:
Where does `np.diff(data).std()/2**.5` come from? Don't just just want `np.std`?

hartytp, 2019-10-28:
Pretty sure sampler currently samples at 500kHz (try injecting a 1kHz sinusoid from an AWG to confirm).

pathfinder49, 2019-10-28:
For a comparison of the no capacitor related noise

hartytp, 2019-10-28:
> For a comparison of the no capacitor related noise

Okay...but I don't get where that formula came from in the first place...

hartytp, 2019-10-28:
nit: always use grids for that kind of plot, and may be best to give y scale as a log (dBV/sqrt(Hz))

hartytp, 2019-10-28:
- did you add a 10nF capacitor in place of the 10uF yet? Or is this all no capacitor?
- is the NSD you're plotting there RTI (i.e. divided by the gain)?

hartytp, 2019-10-28:
I don't understand the data on the plot. AFAICT it says 1uV/rtHz for 500kHz BW. That would be 1.4nV/rtHz. Even if that's RTI it's non-physically low (1kOhm is 4nV/rtHz at room temperature and we have a bunch of 1k resistors. Check the noise model, but I'd guess the AFE noise floor is closer to 10nV/rtHz). So, check it in the AM, but I don't trust your data yet...

dnadlinger, 2019-10-28:
> I don't understand the data on the plot. AFAICT it says 1uV/rtHz for 500kHz BW. 

Wait what? 1 µV / rtHz is 1 µV / rtHz no matter the bandwidth. For, order-of-magnitude 500 kHz noise bandwidth, this gives 0.7 mV rms, so factors of 2 might be wrong, but not factors of 1000.

hartytp, 2019-10-28:
@dnadlinger indeed...it’s late and my brain clearly no longer functions!

hartytp, 2019-10-28:
Still though 1uV/rtHz would be 0.5nVrtHz before gain of 2k, so still at least an order of magnitude too low.

pathfinder49, 2019-10-29:
It was suggested in the stabilizer characterisation

pathfinder49, 2019-10-29:
Working through the circuit diagram, I hadn't accounted for the op amp dividing input voltage by 5. 

I've updated the nsd and voltage above.

hartytp, 2019-10-29:
> Working through the circuit diagram, I hadn't accounted for the op amp dividing input voltage by 5.

So, the calculation you need is that 2^16 is 20V at the input to Stabilizer (the header/SMA).

pathfinder49, 2019-10-29:
Here is the nsd after savgol filtering
[image]

Overall the NSD is better than expected from the noise calculation. [This may have to do with the bandwidth.](https://github.com/OxfordIonTrapGroup/stabilizer_current_sense/issues/12)

hartytp, 2019-10-29:
Thanks!

Assuming the spice model is correct, the 3dB bandwidth of the circuit should be ~75kHz. However, the plot doesn't really look consistent with that. Which is odd.

I assume that the large spike near DC is an artefact of the filter? It's probably worth plotting this kind of thing on a semi-log x scale (or at the very least using kHz as the scale).

Assuming the noise in the pass band is ~16uV/rtHz (eyeballing the plot), before the gain of 2k, that would be 8nV/rtHz, which is still non-physically low...

So, I'm not sure we need to spend more time on this right now, but things don't quite add up yet.

I think a good next step would be to use an AWG to inject a low amplitude sinusoid current pulse into the sense resistor and measure the transfer function through the circuit. That way we can confirm the BW of the various stages.

I think it would also be a good idea to hook it up to the FFT analyzer, which will give you a NSD unit plot. That way we can check your maths.

Anyway, I think this is getting out of the scope of the current investigation, so I suggest you leave it here for now and move on to Thermostat.

pathfinder49, 2019-10-29:
NOTE: my NSDs are a factor 2 smaller than they should be as I misunderstood the definition.

hartytp, 2019-10-30:
@pathfinder49 I uploaded a spice model of the AFE as I was curious to see how good my analytic treatment was. The simulated low frequency noise is approx 11.4nV/rtHz, with a 3dB cut-off around 55kHz. By eye, the bandwidth agrees well with your plot. After fixing that factor of two, you get something like 16nV/rtHz RTI noise, where as the model gives 11.4nV/rtHz. So the measured value is 40% higher. Which is a bit large, but noise measurements never agree too well.

NB the OpAmp noise typical in the data sheet is 2.8nV/rtHz, max is 4.5nV/rtHz, which is 60% difference. In practice I wouldn't expect the variance to be that bad, but I'm also not concerned with this level of discrepancy. So, all in all, it looks like the AFE works! Great!