OxfordIonTrapGroup/stabilizer_current_sense #10: DAC-Filter-Capacitor causes large temp-co

pathfinder49, 2019-10-28:
C33 is causing ~50 mV fluctuations in V_err. 
[image]
Covering the capacitor with tissue removes the fluctuations. De-soldering the capacitor gives a steady V_err with noise consistent with the unfiltered DAC. 

For the de-soldered capacitor:
RMS noise = 8.4 mV
200 kHz pole in V_offset addition
2000x gain to v_err
=> 9 nV/sqrt(Hz) noise
From DAC data-sheet: 11.8 nV/sqrt(Hz)

[image]

cjbe, 2019-10-28:
Good find! What is the plan to fix this - replace it with an electrolytic?

pathfinder49, 2019-10-28:
The Previous measurement was with the boar:
> 
> TS
> 270/19
> 002

Using another board:
> 
> TS
> 270/19
> 001

The behavior is NOT observed. The output is stable with 6 mV RMS noise. Touching the capacitor still causes causes large voltage spikes of over 1 V.

pathfinder49, 2019-10-28:
> Good find! What is the plan to fix this - replace it with an electrolytic?

Seems to be the case of a broken capacitor?

hartytp, 2019-10-28:
I don’t know what to do here. Let’s gather statistics with the other boards before making a decision. Maybe it was a broken cap (I don’t understand the physics behind such a large thermal emf otherwise)? 

With electros I’d worry about the leakage temp co as the dac has a high z output. Could consider reducing the cap size and going for a large plastic?

dnadlinger, 2019-10-28:
We could always add a second buffer stage and use the leakage-free parallel RC technique (i.e. keep zero volts across one capacitor up to R matching, which can then be leaky).

dnadlinger, 2019-10-28:
It really seems like this shouldn't be that big an effect, though.

hartytp, 2019-10-30:
Let's assume it was a broken cap. We'll gather more statistics for looking at other boards, and only worry if the issue surfaces on a new board...

closed 2019-10-30 by hartytp