OxfordIonTrapGroup/stabilizer_current_sense #17: Non-C0G capacitor in frontend offset filter

DougalMain, 2020-02-05:
It's been noticed that the non-C0G capacitor (C33) in the frontend offset filter is very sensitive to thermal fluctuations. This leads to fluctuation of the feedback setpoint and thus causes large, low frequency drifts in the coil supply current.

To characterise this the open loop noise was measured for two (seperate) stabilizers, one with the old non-C0G capacitor and one with a plastic foil capacitor of equal capacitance (10uF).

Blue - with original capacitor
Orange - with replacement capacitor

[image]

While there are other non-C0G capacitors in the circuit, C33 has by far the largest effect. Should either reduce the capacitance of C33 so that a C0G capacitor can be used or find another alternative.

dtcallcock, 2020-02-06:
I've used these reflow-compatible SMT film caps in designs like this where they don't make C0G big enough.  Only downside is that they are $$$ and about 4x the size of an X5R.

https://www.digikey.com/product-detail/en/rubycon/35MU106MD35750/1189-2530-1-ND/4969588

dnadlinger, 2020-02-08:
Thanks for the tip – spendy! We'll probably rejig this part of the circuit instead to use the "bootstrapped" RC filter trick (where 0V is across the top capacitor, which can then have leakage, courtesy of a duplicate RC), although we'll have to add an op-amp buffer after the DAC.

dtcallcock, 2020-02-09:
> We'll probably rejig this part of the circuit instead to use the "bootstrapped" RC filter trick (where 0V is across the top capacitor, which can then have leakage, courtesy of a duplicate RC), although we'll have to add an op-amp buffer after the DAC.

Out of interest, is there an advantage to this other than saving a few $?  Seems it increases complexity, noise, board area, and power consumption.  Admittedly none of them by much.